TcMenu/TaskManagerIO #8: Ability to provide a timer schedule that is greater than 1 hour

davetcc, 2020-08-28:
At the moment, both repeating and fixed tasks have to occur within around 1.193 hours of scheduling.

verysmallrock, 2021-06-04:
I just ran into this with IOAbstraction version 1.7.5 and I think there is a somewhat nasty bug hidden somewhere in here that results in tasks not being run.  

I was trying to add a periodic reset of my LCD code every 12 hours.  My code is like this:
```
taskManager.scheduleFixedRate(220, readAndRedraw);
taskManager.scheduleFixedRate(60000, logData);
taskManager.scheduleFixedRate(2400, updateRelayStates);
taskManager.scheduleFixedRate(2500, exitEditingIfIdle);
taskManager.scheduleFixedRate(5000, updateFans);
taskManager.scheduleFixedRate(60000 * 60 * 12, reinitDisplay);  // unsupported schedule rate
```
([link to my code](https://github.com/verysmallrock/FridgeFermenter/blob/446b47c1e0af31d3827f8dc015de0c73be8cdea8/MiniFridgeFermenter.ino#L64))

After three minutes of my arduino running **only the highest priority task continued to be called** (`readAndRedraw`), which essentially **hangs** my system.  If I commented out the line with `reinitDisplay` everything worked fine.

It's easy to work around this with an extra layer of code that checks `millis()`, e.g. `taskManager.scheduleFixedRate(60000, reinitDisplayCheck);`.  ([code](https://github.com/verysmallrock/FridgeFermenter/blob/34a6daa4d7c9a0b2bed0f35de0a3313d374c9416/Main.cpp#L410))

Anyways, just wanted to leave a note here in case anyone else runs into this.  It'd be great to fix this issue or at least add a warning/error somehow when going over the 1 hour limit.

davetcc, 2021-06-04:
Many thanks for reporting this, it looks like there must be a problem when the schedule is over a certain size, it is probably overflowing the timer type. I will run the above code with a hardware debugger and find the exact cause of the freeze as a priority. 

At the minimum, if the schedule is too large, we should probably not create the task and return the invalid task id.

davetcc, 2021-06-04:
Our longer-term plan for this is on the first very long schedule, EG over a few minutes, to start a watchdog task that monitors the longer tasks and schedules them as they become ready for execution. These would be scheduled within a few seconds of their intended time basically. Which when the interval is in hours, is probably close enough for most. 

They would not be in the highly efficient time-based linked queue, but given they would only be checked every few seconds, that should not matter too much. This would allow very large schedules at around a second accuracy. In fact we could even provide a compiler flag that set the accuracy in milliseconds, defaulting to maybe 2500 millis check frequency for long-running tasks over say 30 minutes.

davetcc, 2021-06-04:
In fact with this kind of task, rethinking it produces a better solution. It can be implemented as an event instead of a task, events have variable timing, and can easily be triggered early (in the case of a new task arriving). So we can make it wake up very infrequently unless there's something to do. Adding a new long-running task would just trigger the event immediately where it could re-evaluate everything.

verysmallrock, 2021-06-05:
Oh, I wasn't aware of Events but my workaround does seem very similar -- i.e. check every minute, but only execute my task every 12 hours.

A quick thought: I really like the abstraction that `taskManager.scheduleFixedRate` provides since it's so easy to understand, integrate, and use.  It 'sells well' when you're Googling for libraries and I expect you have a lot of beginners using it to help make code more readable and maintainable. That probably also means you have a lot of users keeping their integration pretty simple (like me), so if `scheduleFixedRate` were to just switch to a less-accurate mode if given a large timeout, that would totally work for me. Accuracy within a minute would be fine for my purposes.

davetcc, 2021-06-05:
Sorry no the event fix is for me to implement the long task support basically, so that you could just schedule as normal even tasks scheduled days away.

Hawkeye4040, 2021-08-28:
Yeah, I really like that approach much better than what I was thinking of. There may still be a little benefit with what I was thinking but it's more of a band-aid patch in comparison to that.

I'm sure it's obvious that I have a big interest in asynchronous programming in embedded and well basically anything that'll squeeze as much performance as we can get out of the chips we have available now. They are getting better but the main thing is we can much better utilize the resources at our disposal now and I think this approach is going to help immensely with that.

davetcc, 2021-08-29:
> There may still be a little benefit with what I was thinking but it's more of a band-aid patch in comparison to that.

The problem is that it would massively affect performance on a wide range of boards. Most boards do not have native 64-bit support, and even if they do, AFAIK there are very few true 64 bit processors in the embedded domain, especially ones task manager targets.

Further, there are some operations that need to be atomic (even with CAS), IE a guarantee that what we see is truly the latest, and what we write is safely replacing what came before it. These operations often do not even exist for 64-bit types, let alone on 8-bit, even on many 32-bit boards.

> I'm sure it's obvious that I have a big interest in asynchronous programming in embedded 

That's great, and it's good to have people such as yourself along for the ride! Another advantage of task manager is to make life easier for programmers.

Hawkeye4040, 2021-08-30:
Ah, bummer there's always something like that that you don't see at first. That is a lot of extra work to satisfactorily support 64-bit then! Yeah, there are some 64-bit MPU's that I use but it's quite a pain on the programming side due to immature API support. I was also messing around with Master-slave configurations with 2x 32-bit PIC and ARM MCU's and you see the bottlenecks you'd expect to see.

I'll do more research but I really want to see the upcoming enhancements you mentioned first as I know that'll change how I approach this too. I appreciate that heads up there so I'm not backtracking my work there as much.

davetcc, 2021-10-03:
Actually, what I expected to be many hours or even days of effort turned out to have a very simple solution. Testing it now..

1/ We create a schedule object globally

    TmLongSchedule hourAndHalfSchedule(makeHourSchedule(1, 30), [] { workToDo });

2/ We register it with taskManager

    taskManager.registerEvent(&hourAndHalfSchedule);

Or even if you prefer using new 

    taskManager.registerEvent(new TmLongSchedule(makeHourSchedule(0, 15), [] {
        Serial.print(millis());
        Serial.print(": Fifteen minutes passed");
    }), true);