TcMenu/TaskManagerIO #16: Problems with writing to file

stefunkk, 2020-11-05:
Hi

I have task which saves the data to file, it's a logger:

https://github.com/stefunkk/OpenStill/blob/master/src/StillDataTask.cpp

Which saves the data using SPIFFS:

https://github.com/stefunkk/OpenStill/blob/master/src/FileService.cpp

File file = _fs.open(path, "a");
file.print(content);
file.close();

But I'm getting strange behavior inside a file, instead of having nice structure line by line sometimes it adds strange characters at the end, and everything is mixed up, like here:

22.500000;22.500000;22.750000;22.687500;яяя22.500000;22.500000;22.750000;22.687500;0
22.500000;22.500000;22.750000;22.687500;яяя22.500000;22.500000;22.750000;22.687500;0
22.500000;22.500000;22.750000;22.625000;0
22.500000;23.000000;22.750000;22.625000;яяя22.500000;22.500000;22.750000;22.625000;яяя

Is it possible that this behavior is caused by asynchronous processing?

davetcc, 2020-11-05:
Please could you raise issues with isolated examples, when you point us to a large project we cannot help you to debug that. We have already provided you a lot of assistance in a prior incident in LiquidCrystalIO. We are a small company and cannot keep providing you this level of coding assistance free of charge. 

Whenever writing anything asynchronous, you need to consider how the data is shared, and how you will avoid things being overwritten, deleted, etc. This is part of writing any large solution, especially in C++.

Is this even a valid format string with an un-escaped % in it? I would look there first.

```%f;%f;%f;%f;%i%\n```

stefunkk, 2020-11-05:
Dave I really appreciate your help and understand your point of view. If you send me paypal link I can buy you a beer for wasting time on previous issue ... i feel really bad about it.

I will try to keep examples small for future, hope that helps.

https://pastebin.com/in9vWs92

My results:

(...begining was pretty ok, but after 20 sec...)
0;0;0;0;���0;0;0;0;���0;0;0;0;0
0;0;0;0;0
0;0;0;0;���0;0;0;0;���0;0;0;0;0

stefunkk, 2020-11-05:
If i flash same code with arduino instead of platformio problem disappear

stefunkk, 2020-11-05:
Code without TaskManagerIO library still broken on PlatformIO, so for sure it's not a TaskMangagerIO fault. My bad again. You can close the bug.

https://pastebin.com/U0SGbEiA

davetcc, 2020-11-05:
Thanks for letting me know. Don't worry about the last one BTW. 

Line 63 Serial.println(fileRead.readString());

Looks deeply suspicious somehow, who owns that readString memory, it's probably passed to you by reference, and that ref may be gone by the time you use it, maybe try reading into a char array (such as char[200]) and see if that fixes it

closed 2020-11-05 by davetcc