TcMenu/TaskManagerIO #21: compilation error for SAMD21

Ulli2k, 2020-12-23:
Can not be compiled. I get the following error on SAMD21
```
In file included from .pio/libdeps/zeroUSB/TaskManagerIO/src/TaskManagerIO.cpp:7:0:
.pio/libdeps/zeroUSB/TaskManagerIO/src/TaskManagerIO.h: In member function 'virtual void BasicArduinoInterruptAbstraction::attachInterrupt(pintype_t, RawIntHandler, uint8_t)':
.pio/libdeps/zeroUSB/TaskManagerIO/src/TaskManagerIO.h:86:40: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'PinStatus' [-fpermissive]
         ::attachInterrupt(pin, fn, mode);
```

```
 from .pio/libdeps/zeroUSB/TaskManagerIO/src/TaskManagerIO.cpp:6:
/home/ulli/.platformio/packages/framework-arduino-samd/cores/arduino/api/Common.h:111:6: note:   initializing argument 3 of 'void attachInterrupt(pin_size_t, voidFuncPtr, PinStatus)'
 void attachInterrupt(pin_size_t interruptNumber, voidFuncPtr callback, PinStatus mode);
      ^~~~~~~~~~~~~~~
```

davetcc, 2020-12-23:
We are looking at this now, should have a fix very soon.

davetcc, 2020-12-23:
Many thanks for reporting this. This actually needs a bit of thought, because the Arduino core has been changed in a way that has not been propagated to other cores such as seeed for example. I'm looking at it now to work out how to handle this. What I've done for now is move this BasicArduinoInterruptAbstraction class into its own file, as most people don't use it, and in new code it would be better to use the BaseEvent for interrupt handling instead.

Many users would use this with an IoAbstraction, and that IS broken too, It's going to need some careful thought to work out how this will work.

For now, just change the class in taskmanagerio.h to look as follows while I think this through:

```class BasicArduinoInterruptAbstraction : public InterruptAbstraction {
    void attachInterrupt(pintype_t pin, RawIntHandler fn, uint8_t mode) override {
#if defined(ARDUINO_MBED_MODE)
        ::attachInterrupt(pin, fn, (PinStatus)mode);
#elif defined(PARTICLE)
        ::attachInterrupt(pin, fn, (InterruptMode)mode);
#elif defined(ARDUINO_ARCH_SAMD)
        ::attachInterrupt(pin, fn, (PinStatus)mode);
#else
        ::attachInterrupt(pin, fn, mode);
#endif // ARDUINO_MBED_MODE
    }
};```

davetcc, 2020-12-24:
Found it, seems a breaking change was made in Arduino Core API. Luckily we can detect it easily enough. Change will be released later today.

davetcc, 2020-12-24:
See https://github.com/arduino/ArduinoCore-samd/issues/587 where I've questioned some of the changes in this core.

davetcc, 2020-12-24:
I am testing a fix on master at the moment, it is difficult as even some quite core libraries such as AUnit that we use for unit testing don't work with this core!

davetcc, 2020-12-24:
There is a fix for this on master that is undergoing testing. I recommend for most users to revert their SAMD board to the 1.8.9 until I have fully tested the change on all boards.

davetcc, 2020-12-24:
We now support both variants of SAMD, core API and regular. Please re-open if it doesn't work for you in task manager.

closed 2020-12-24 by davetcc