TcMenu/TaskManagerIO #22: Incompatibility with SoftwareSerial / ESP8266

rhscdn, 2021-02-09:
Are there any known incompatibilities between TaskmanagerIO and SoftwareSerial in the arduino environment? I'm trying to run TaskManagerIO on a Wemos D1 Mini (R2). I'm seeing definitions clashing and unexpected behaviour.

NB: I should also say that I'm new to TaskManagerIO.

**If I include TaskManagerIO.h before SoftwareSerial.h I get:**

```Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - tool-mklittlefs 1.203.210203 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TaskManagerIO> 1.1.1
|-- <EspSoftwareSerial> 6.8.5
Building in release mode
Compiling .pio/build/d1_mini/src/activeTest2.cpp.o
In file included from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskPlatformDeps.h:86:0,
                 from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskManagerIO.h:9,
                 from src/activeTest2.cpp:3:
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:89:0: warning: "ATOMIC_VAR_INIT" redefined [enabled by default]
 #define ATOMIC_VAR_INIT(value)  { .__val = (value) }
 ^
In file included from /home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:41:0,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:28,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/SoftwareSerial.h:27,
                 from src/activeTest2.cpp:2:
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/atomic_base.h:234:0: note: this is the location of the previous definition
 #define ATOMIC_VAR_INIT(_VI) { _VI }
 ^
In file included from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskPlatformDeps.h:86:0,
                 from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskManagerIO.h:9,
                 from src/activeTest2.cpp:3:
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:380:0: warning: "ATOMIC_FLAG_INIT" redefined [enabled by default]
 #define ATOMIC_FLAG_INIT  { ATOMIC_VAR_INIT(0) }
 ^
In file included from /home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:41:0,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:28,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/SoftwareSerial.h:27,
                 from src/activeTest2.cpp:2:
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/atomic_base.h:268:0: note: this is the location of the previous definition
 #define ATOMIC_FLAG_INIT { 0 }
 ^
Retrieving maximum program size .pio/build/d1_mini/firmware.elf
Checking size .pio/build/d1_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.3% (used 28076 bytes from 81920 bytes)
Flash: [===       ]  26.1% (used 272164 bytes from 1044464 bytes)
```

**If I reverse the order and include SoftwareSerial.h before TaskManagerIO.h, the build fails:**
...
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
[... 39 lines cut ...]
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:90:36: error: expected ')' before 'void'
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:284:18: error: variable or field '__atomic_store_n' declared void
  __atomic_store_n(&(object)->__val, desired, order)
                  ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '*' token
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: '__a' was not declared in this scope
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '__i'
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '__m'
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:284:18: error: variable or field '__atomic_store_n' declared void
  __atomic_store_n(&(object)->__val, desired, order)
                  ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:828:5: error: expected primary-expression before 'volatile'
     atomic_store_explicit(volatile atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:828:5: error: expected ')' before 'volatile'
*** [.pio/build/d1_mini/src/activeTest2.cpp.o] Error 1
...

davetcc, 2021-02-09:
Yes it looks like software serial and task manager are using std::atomic to provide atomic access to variables.

We test every release with ESP8266, but there must be a conflict in libraries in software serial. We'll need to try this locally to see what the incompatibility is and feed back.

rhscdn, 2021-02-09:
Thanks, that was my interpretation as well.  I'm new to TaskManagerIO and didn't dig too deep. Let me know if I can provide any other info.

davetcc, 2021-02-15:
recreated and fixing. We will move to the new method of using atomic, that is

#include <atomic>

We'll switch to using the direct compare and exchange function too. Should be in the repo shortly once coded up and tested.

davetcc, 2021-02-15:
I've committed a potential fix to master, found two issues, first the include was inside a namespace, second, we were using an older way to include atomic, fixed both potentially. Tested best I could locally, could you feed back if it works for you. I'll do a test on ESP32, AVR and MKR to get more confidence and then re-release.

davetcc, 2021-02-17:
Tested on ESP8266, ESP32, SAMD and AVR. Releasing.

closed 2021-02-17 by davetcc