TcMenu/TaskManagerIO #31: How I can  add this flag to my compile options esp8266

mrdsn3, 2021-06-30:
How I can  add this flag to my compile options  DTM_ENABLE_CAPTURED_LAMBDAS

davetcc, 2021-07-01:
Depends on the IDE and build tool basically. 

You need to add a compiler define flag. For example in platformIO you add build_flags to a target as per: https://github.com/davetcc/tcLibraryDev/blob/master/platformio.ini

While regular Arduino IDE does not support build flags directly, you would need to manually set it in the file.

davetcc, 2021-07-01:
BTW the flag is TM_ENABLE_CAPTURED_LAMBDAS the -D at the beginning is the option for setting a flag in gcc.

mrdsn3, 2021-07-01:
thanks  so is this code added the flag? is this right?
`#include <TaskManagerIO.h>
#if !defined(TM_ENABLE_CAPTURED_LAMBDAS)
# define TM_ALLOW_CAPTURED_LAMBDA
#endif
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

davetcc, 2021-07-01:
Unfortunately for Arduino IDE it has to go into TaskPlatformDeps.h, otherwise it will not be picked up when compiling the library.

If you are using the pro IDE, Eclipse, or platformIO you can define the flag in the build tool.

mrdsn3, 2021-07-01:
OK thanks

closed 2021-07-01 by mrdsn3