TcMenu/TaskManagerIO #44: One time event (long schedule)

lablabla, 2022-10-22:
Hi,

I have a need for a `TmLongSchedule` but it should only run once.
Is there a reason for not having such option?

Currently I'm thinking of doing something like this. Storing the created task id, and inside the lambda function, retrieve it and cancel the task

```
auto taskId = taskManager.registerEvent(new TmLongSchedule(next_run*1000, [this, &event]() {    
    taskid_t taskIdToCancel = this->getTaskId(event.id);
    taskManager.cancelTask(taskIdToCancel);

    // do task after long wait
}), true);

if (taskId == TASKMGR_INVALIDID)
{
    log_e("Failed scheduling task");
    return;
}
m_jobs[event.id] = taskId;
```

It seems to work but it seems like a relatively hacky way of doing this, considering for "short" schedule there exists a `scheduleOnce`

davetcc, 2022-10-22:
Examine if this is fairly straightforward and add to the current release if so.

davetcc, 2022-10-25:
Actually, this literally turns out to be a oneliner, I've added a defaulted extra parameter to the constructor called `oneTime`

davetcc, 2022-10-25:
Will be released as soon as testing is concluded on a range of boards.

closed 2022-10-25 by davetcc

davetcc, 2022-10-26:
https://www.thecoderscorner.com/products/arduino-libraries/taskmanager-io/task-manager-scheduling-guide/#scheduling-tasks-more-than-an-hour-in-duration