TcMenu/TaskManagerIO #53: 1.4.1 compile problem for nano esp32

tomelgato, 2024-04-15:
On PlatformIO with board "arduino_nano_esp32" compile fails:

/Users/tm/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:69: error: 'digitalPinToGPIONumber' is not a type
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)

tomelgato, 2024-04-15:
[env:arduino_nano_esp32]
platform = espressif32
board = arduino_nano_esp32
framework = arduino

davetcc, 2024-04-16:
I'm afraid I don't have one of those boards so for now can't fix it. If you can find what's wrong and raise a PR I'm happy to merge.

davetcc, 2024-04-16:
My take would be if a conditional compile for ESP32 remove digitalPinToGPIONumber as it's not needed on ESP32. Seems like more woes from incompatible APIs.

#define attachInterrupt(pin, fcn, mode) attachInterrupt(pin, fcn, mode)

tomelgato, 2024-04-16:
You dont need the board, you just need to set the environment and try to compile.

out of memory ... the Arduino nano esp33 libs are missing the digitalPinToGPIONumber function, i did read somewhere that a user managed to get it running with a generic esp32 library (on arduino ide)

davetcc, 2024-04-16:
More Arduino incompatibilities unfortunately. If you thought things were a mess in the past, at least then writers of cores tried to keep critical functions that were relied on by libraries, now it seems to be a free for all. I'm not sure it is supportable with that missing. **Just to be clear, that is not in code I can do anything about, we are just calling attachInterrupt(pin, function, mode)**.

tomelgato, 2024-04-16:
See here for a solution:

https://github.com/espressif/arduino-esp32/issues/9150

tomelgato, 2024-04-16:
Other solution could be this: 

I tried the IDE Tools option "Pin numbering: By GPIO number (legacy)" and the compilation was OK!

But i dont know if its possible to do in PlatformIO

tomelgato, 2024-04-16:
I didnt post the complete compiler error, the first one is this and then a lot more with the same problem:

Compiling .pio/build/arduino_nano_esp32/lib7f3/SimpleCollections/SCThreadingSupport.cpp.o
Compiling .pio/build/arduino_nano_esp32/lib7f3/SimpleCollections/SimpleCollections.cpp.o
In file included from /Users/tm/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:223,
                 from .pio/libdeps/arduino_nano_esp32/TaskManagerIO/src/TaskPlatformDeps.h:28,
                 from .pio/libdeps/arduino_nano_esp32/TaskManagerIO/src/TaskManagerIO.cpp:6:
/Users/tm/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:69: error: 'digitalPinToGPIONumber' is not a type
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/arduino_nano_esp32/TaskManagerIO/src/TaskManagerIO.h:64:18: note: in expansion of macro 'attachInterrupt'
     virtual void attachInterrupt(pintype_t pin, RawIntHandler fn, uint8_t mode) = 0;

tomelgato, 2024-04-16:
> More Arduino incompatibilities unfortunately. If you thought things were a mess in the past, at least then writers of cores tried to keep critical functions that were relied on by libraries, now it seems to be a free for all. 

The pin numbering puzzled me some times already. The convention for the ESP using D2 instead of 5 ... dont know if i like it or not. If it would be across all Arduino boards it could be good. Btw. with the FastAccelStepper library as example i had to use the #5 instead of D2 ... so confusion will stay ;)

davetcc, 2024-04-17:
I've commented on the thread above, they've broken the core, and many libraries will not compile on it. Unfortunately, my take is to use a board that works. I doubt a fix will come along any time soon.

me-no-dev, 2024-04-17:
Hi @tomelgato . I am no platformio fluent, but I know there is a way for you to define additional build flags. If you add the following define to your build env, you should be able to compile with pin remap off: `-D BOARD_USES_HW_GPIO_NUMBERS`

tomelgato, 2024-04-17:
@me-no-dev Supercool, thanks a lot. It compiles with that option. Where did you find that info? Even googeling with "BOARD_USES_HW_GPIO_NUMBERS" brings nothing.

me-no-dev, 2024-04-17:
Hi @tomelgato :) Found it in the Arduino board definition (that menu option that allows you to turn it off): https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L35586

baender, 2024-12-14:
Hi there,

first of all, thanks a lot for providing this library!

Second, I also stumbled across this issue today, as I also use an Arduino Nano ESP32.

With the workaround of adding `BOARD_USES_HW_GPIO_NUMBERS` as a build flag, I could compile the project (with platformio). If I understood it correctly, I will need to use the GPIO numbers from here on for all functionality.

I wanted to leave this comment, to stay informed whether there will be a proper solution at some point in the future.

Thanks again @davetcc

davetcc, 2025-10-20:
**In summary**, Arduino team decided to do GPIO numbers a completely different way to all other ESP32 boards, effectively putting work onto every library developer out there. They didn't seem to consider that it would break many existing libraries (see the linked issue in above comments). There isn't a quick fix.

I'm not sure what proper solution we can provide quite honestly as they've changed the way pin mapping works just on that board. If you look at the Adafruit IO expander offering, they have the same issue as well - https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/issues/100

Unfortunately, I don't think there's any fix.

I quote from https://github.com/espressif/arduino-esp32/issues/9150: "choosing By GPIO number (legacy) in the IDE menus
will fix all issues, as in that case no macros are defined at all."