TcMenu/TaskManagerIO #59: TcMenu with VSCode + ESP-IDF v5.1.3 for ESP32 an ST7735 1,8" TFT display

ONik93, 2024-08-18:
Hi,

I've build TcMenu with VSCode + ESP-IDF v5.1.3 for ESP32 an ST7735 1,8" TFT display.

To achieve it i changed this lines in TaskManagerIO and SimpleCollections lib
```
#include "Utils.h"

inline bool casAtomic(position_ptr_t ptr, position_t expected, position_t newVal) {
    uint32_t exp32 = expected;
    uint32_t new32 = newVal;
    // uxPortCompareSet(ptr, exp32, &new32);
    uxPortCompareSetCustom(ptr, exp32, &new32);
    return new32 == expected;
}
```

```
// Utils.h
#include <stdint.h>

void uxPortCompareSetCustom(volatile uint32_t *addr, uint32_t compare, uint32_t *set);
```

```
// Utils.cpp
#include "Utils.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void uxPortCompareSetCustom(volatile uint32_t *addr, uint32_t compare, uint32_t *set) {
    uint32_t prev;
    //BaseType_t taskState;
    int taskState;

    // Enter critical section to ensure atomicity
    taskState = taskENTER_CRITICAL_FROM_ISR();

    prev = *addr;
    if (prev == compare) {
        *addr = *set;  // Update memory location with the new value
    }
    *set = prev;  // Update the 'set' pointer with the previous value

    // Exit critical section
    taskEXIT_CRITICAL_FROM_ISR(taskState);
}
```

And IoAbstraction library required following modification
```

#include <Arduino.h>

void EspAnalogOutputMode::pinSetup() {
    if(!isDac()) {
        // for other than the dac ports, we need to set up PWM
        ledcSetup(pwmChannel, pwmWidth, 8);
        ledcAttachPin(pin, pwmChannel);
        // ledcSetup(pwmChannel, pwmWidth, 8);
        // ledcAttachPin(pin, pwmChannel);
        ledcAttach(pin, pwmWidth, 8);
    }
```

I hope you can implement this fixes to your project.

davetcc, 2024-09-15:
I'm not sure why the port macro uxPortCompareSet from freertos/portmacro.h  doesn't work to be honest, it should really be there as it's part of their core rtos implementation. 

I think the ledc functions may be part of Arduino to be honest and they may need to be fully moved to use IDF functions instead.

We do plan to fully support IDF, but for now, you should be able to get uxPortCompareSet working with IDF, and you could just comment out the ledc functions if Arduino library is not there.

mathansen, 2025-07-22:
Apparently, `uxPortCompareSet` has been removed in IDF v5 last year. Compare

https://github.com/espressif/esp-idf/blob/v5.5/components/freertos/FreeRTOS-Kernel/portable/linux/include/freertos/portmacro.h

with 

https://github.com/espressif/esp-idf/blob/v4.4.7/components/freertos/port/xtensa/include/freertos/portmacro.h#L404

So TaskManagerIO is broken for ESP using current versions, hence the workaround mentioned by @ONik93 is required.

davetcc, 2025-10-18:
I am just doing a few ESP32 smoke tests then I'll do a release.